sibradzic/amdgpu-clocks #5: Vega 56 voltages might not apply

FKleinebreil, 2019-10-08:
I'm using an Asus Vega 56 Strix with Ubuntu Budgie 19.04. I wanted to use you tool (which is nevertheless great btw!) to undervolt the card and overclock the memory. I have the suspicion that while clocks apply, voltages don't.

My custom power states (core clocks as they were + custom voltages (1.2V- -> 1.0V for P7), overclocked P3 memory):
_OD_SCLK:
0:        852Mhz        800mV
1:        991Mhz        900mV
2:       1138Mhz        906mV
3:       1269Mhz       912mV
4:       1312Mhz       918mV
5:       1474Mhz       975mV
6:       1538Mhz       987mV
7:       1590Mhz       1000mV
OD_MCLK:
0:        167Mhz        800mV
1:        500Mhz        800mV
2:        700Mhz        900mV
3:        940Mhz        975mV
OD_RANGE:
SCLK:     852MHz       2400MHz
MCLK:     167MHz       1500MHz
VDDC:     800mV        1000mV_

_/sys/class/drm/cardX/device/pp_od_clk_voltage_ is identical to _/etc/default/amdgpu-custom-state.card0_ after running the script except:
_VDDC:     800mV        1200mV_
1200mV was the default value, too.
When I use WattmanGTK to monitor the Vega 56 during Unigine Superposition the reported vddgfx is 1.2V. Also the GPU temperature hits 80°C, which should not happen if 1.0V were actually applied. The memory clock is read correctly at 940MHz.

Do you have any idea what is going on? Also, is it correct that _/sys/class/drm/cardX/device/pp_od_clk_voltage_ is overwritten by running the script?

Many thanks in advance!

FKleinebreil, 2019-10-08:
Thank you for the fast reply! I removed the _OD_RANGE:_ constraints from _amdgpu-custom-states.card0_.

1. `Linux Florian-PC 5.0.0-31-generic #33-Ubuntu SMP Mon Sep 30 18:51:59 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`

2. `OD_SCLK:
0:        852Mhz        800mV
1:        991Mhz        900mV
2:       1138Mhz        950mV
3:       1269Mhz       1000mV
4:       1312Mhz       1050mV
5:       1474Mhz       1100mV
6:       1538Mhz       1150mV
7:       1590Mhz       1200mV
OD_MCLK:
0:        167Mhz        800mV
1:        500Mhz        800mV
2:        700Mhz        900mV
3:        800Mhz        950mV
OD_RANGE:
SCLK:     852MHz       2400MHz
MCLK:     167MHz       1500MHz
VDDC:     800mV        1200mV`

3. `OD_SCLK:
0:        852Mhz        800mV
1:        991Mhz        900mV
2:       1138Mhz        906mV
3:       1269Mhz       912mV
4:       1312Mhz       918mV
5:       1474Mhz       975mV
6:       1538Mhz       987mV
7:       1590Mhz       1000mV
OD_MCLK:
0:        167Mhz        800mV
1:        500Mhz        800mV
2:        700Mhz        900mV
3:        940Mhz        975mV`

4. `Detecting the state values at /sys/class/drm/card0/device/pp_od_clk_voltage:
  SCLK state 0: 852Mhz, 800mV
  SCLK state 1: 991Mhz, 900mV
  SCLK state 2: 1138Mhz, 950mV
  SCLK state 3: 1269Mhz, 1000mV
  SCLK state 4: 1312Mhz, 1050mV
  SCLK state 5: 1474Mhz, 1100mV
  SCLK state 6: 1538Mhz, 1150mV
  SCLK state 7: 1590Mhz, 1200mV
  MCLK state 0: 167Mhz, 800mV
  MCLK state 1: 500Mhz, 800mV
  MCLK state 2: 700Mhz, 900mV
  MCLK state 3: 800Mhz, 950mV
  Maximum clocks & voltages:
    SCLK clock 2400MHz
    MCLK clock 1500MHz
    VDDC voltage 1200mV
  Curent power cap: 260W
Verifying user state values at /etc/default/amdgpu-custom-state.card0:
  SCLK state 0: 852Mhz, 800mV
  SCLK state 1: 991Mhz, 900mV
  SCLK state 2: 1138Mhz, 906mV
  SCLK state 3: 1269Mhz, 912mV
  SCLK state 4: 1312Mhz, 918mV
  SCLK state 5: 1474Mhz, 975mV
  SCLK state 6: 1538Mhz, 987mV
  SCLK state 7: 1590Mhz, 1000mV
  MCLK state 0: 167Mhz, 800mV
  MCLK state 1: 500Mhz, 800mV
  MCLK state 2: 700Mhz, 900mV
  MCLK state 3: 940Mhz, 975mV
Committing custom states to /sys/class/drm/card0/device/pp_od_clk_voltage:
  Done`

5. `OD_SCLK:
0:        852Mhz        800mV
1:        991Mhz        900mV
2:       1138Mhz        906mV
3:       1269Mhz        912mV
4:       1312Mhz        918mV
5:       1474Mhz        975mV
6:       1538Mhz        987mV
7:       1590Mhz       1000mV
OD_MCLK:
0:        167Mhz        800mV
1:        500Mhz        800mV
2:        700Mhz        900mV
3:        940Mhz        975mV
OD_RANGE:
SCLK:     852MHz       2400MHz
MCLK:     167MHz       1500MHz
VDDC:     800mV        1200mV`

However, WattmanGTK still reports a vddgfx (whcih I assume is core voltage) of 1.2V and temps reach 80°C under load quickly, as I said. Or to say it differently: Temps are identical before and after after the change.

sibradzic, 2019-10-08:
Can you please add proper newlines to those outputs in the comment above, I can't read a thing like this...

sibradzic, 2019-10-08:
> However, WattmanGTK still reports a vddgfx (which I assume is core voltage) of 1.2V and temps reach 80°C under load quickly, as I said. Or to say it differently: Temps are identical before and after after the change.

I have no clue about WattmanGTK, just to be sure it is sane, compare the output of (as root)
`watch -n1 "cat /sys/kernel/debug/dri/0/amdgpu_pm_info | tail -n16"` (keep the terminal "always on top" and compare MCLK, voltage, wattage & temperature outpus)

FKleinebreil, 2019-10-08:
Sorry, this is my first time using the github forum. There were proper new lines before the `insert code` and I don't know how to add newlines in that environment. I'll google it and edit it.

I cross checked with the command you gave me and it too reports a VDDGFX of 1200mV, while the changed memory voltage (940mV) seems to be applied correctly.

sibradzic, 2019-10-08:
Thanks for the output fix! According to the output, noting wrong with the script itself, the clocks and voltages are being set correctly. As of the VDDGFX report, it is indeed strange, it could be related to requirement to set the clock and voltages in order (see https://github.com/RadeonOpenCompute/ROCm/issues/463 for details). Perhaps you can test applying the following `/etc/default/amdgpu-custom-states.card0`:
```
OD_SCLK:
7:       1590Mhz       1170mV
OD_MCLK:
3:        800Mhz        930mV
FORCE_SCLK: 7
FORCE_MCLK: 1
FORCE_PERF_LEVEL: manual
```
reboot before you try, and check the output of `cat /sys/kernel/debug/dri/0/amdgpu_pm_info | tail -n16` before and after you apply the custom states (no need to generate GPU load, both SCLK & MCLK states are being forced here)...

FKleinebreil, 2019-10-08:
I edited the power states you suggested, rebooted and apllied them. this is the output:
```
GFX Clocks and Power:
	800 MHz (MCLK)
	1633 MHz (SCLK)
	1269 MHz (PSTATE_SCLK)
	700 MHz (PSTATE_MCLK)
	1200 mV (VDDGFX)
	37.0 W (average GPU)

GPU Temperature: 48 C
GPU Load: 0 %

SMC Feature Mask: 0x000000001ba1fb4f
UVD: Disabled

VCE: Disabled
```
So the issue remains. However, I get that the issue is not with your script but somewhere else. So there is probably not much you can do about it. I will see if I can find someone with similar issues.

Thank you very much for your time!

closed 2019-10-08 by FKleinebreil

sibradzic, 2019-10-09:
If you really applied that custom state I suggested correctly, that output of `amdgpu_pm_info` is super strange indeed. Are you sure you don't have some other over/under/clocking/volting software active at the same time? Do you perhaps have multiple graphic cards in your system?

FKleinebreil, 2019-10-23:
I don't have other software active and only a single Vega 56. No iGPU either, CPU is a Ryzen 5 3600.

sibradzic, 2019-10-31:
This seems to be the bug affecting you, and apparently the fix was just applied to 5.4 RC:
https://bugs.freedesktop.org/show_bug.cgi?id=109887
https://bugzilla.kernel.org/show_bug.cgi?id=205277

FKleinebreil, 2019-11-06:
Thank you very much for the Update!

xcom169, 2022-04-14:
I have 5.15.28 kernel, but I think I have the same issue with Vega56