sibradzic/amdgpu-clocks #6: Completed install.  Flickering after login.

dmbomer, 2019-10-25:
System setup
AMDFX8230
32GB RAM
240GB primary

MSI RX570 8GB Armour DP ports, using DP to HDMI cables
3 monitors setup Benq GW2280

Ubuntu 18.04.3 fresh install
Linux deepcool 5.0.0-31-generic #33~18.04.1-Ubuntu SMP Tue Oct 1 10:20:39 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
AMDGPU 19.30 is installed
./amdgpu-install --opencl=legacy

After I set the amdgpu.appfeaturemask in grub, update grub and reboot..Done

GDM3 no longer flickers
After I login the screen starts flickering.  

After I run sudo amdgpu-clocks in the terminal the flickering stop.  

Where am I going wrong here?

sibradzic, 2019-10-25:
This is a multi-monitor issue, not related to `amdgpu-clocks`. See https://www.phoronix.com/scan.php?page=news_item&px=AMDGPU-Multi-Monitor-vRAM-Clock. The way AMD driver (on windows as well?) mitigates the issue is to stick to highest MCLK state all the time, hence increasing the idle power usage. It seems that enabling OverDrive via *appfeaturemask* somehow overrides static MCLK multi-monitor workaround in amdgpu driver...
Setting *FORCE_MCLK: 2* in `amdgpu-clocks` (followed by `FORCE_PERF_LEVEL: manual`) essentially does the same workaround as above, freezes your MCLK to highest state, so flicker goes away...

dmbomer, 2019-10-26:
Hi Samir
I don't have windows installed on this system.
I will check it out the article
Thank you for the information Sir

David

closed 2019-10-27 by sibradzic