sibradzic/amdgpu-clocks #7: Add power profile management

ittomsk, 2019-12-18:
Add power profile management

sibradzic, 2019-12-18:
Hi @ittomsk 
This looks good, you want to submit it as PR?

ittomsk, 2019-12-19:
@sibradzic yes. will be great

sibradzic, 2019-12-19:
cool. BTW, I did little driver reading last night, there is a way to set custom power profile;
https://github.com/torvalds/linux/blob/master/drivers/gpu/drm/amd/amdgpu/amdgpu_pm.c#L1230
So, something like this would work;
`echo "5 1 0 100 30 1 0 100 50" > /sys/class/drm/card0/device/pp_power_profile_mode`
I'd validate FORCE_POWER_PROFILE in a bit different manner if I was you...

ittomsk, 2019-12-19:
Well, that was a quick and dirty "copy and modify" hack. So if you think there is a better way of doing it, just do it. May be you'll add custom profile as well. Than will be great too

commit cfa82cab (2020-01-15): Add support for chosing pp_power_profile_mode (fixes #7)
closed 2020-01-15 by sibradzic