sibradzic/amdgpu-clocks #9: Script fails on 5700xt using Kernel 5.5.5

Keridos, 2020-02-25:
First of all the voltages in VDDC Curve are not correctly read since the kernel interface has an "@" added between clcok and voltage. For that to display correctly the function that shows that just needs to parse the 4th word on the line, not the third. Also applies to the veryfying user state values. it ommits the voltage completely from there.

This is how the pp_od_clk_voltage looks like on my card:
```
OD_SCLK:
0: 800Mhz
1: 1800Mhz
OD_MCLK:
1: 875MHz
OD_VDDC_CURVE:
0: 800MHz @ 709mV
1: 1350MHz @ 790mV
2: 1800MHz @ 1100mV
OD_RANGE:
SCLK:     800Mhz       2150Mhz
MCLK:     625Mhz        950Mhz
VDDC_CURVE_SCLK[0]:     800Mhz       2150Mhz
VDDC_CURVE_VOLT[0]:     750mV        1200mV
VDDC_CURVE_SCLK[1]:     800Mhz       2150Mhz
VDDC_CURVE_VOLT[1]:     750mV        1200mV
VDDC_CURVE_SCLK[2]:     800Mhz       2150Mhz
VDDC_CURVE_VOLT[2]:     750mV        1200mV

```

sibradzic, 2020-02-26:
Thanks for report. I've just booted 5.6 RC3 and I can finally see the similar output on my 5700. I guess the pp_od_clk_voltage kernel interface went through some transitions lately, I'll try to adjust the parsing to match accordingly as soon as I get some free time...

sibradzic, 2020-02-27:
@Keridos fixed, can you please confirm?

sibradzic, 2020-03-07:
@Keridos ping

closed 2020-03-08 by sibradzic

Keridos, 2020-04-09:
Sorry for answering this late: currently I have this error output when trying to get  custom state to work: 
https://gist.github.com/Keridos/4b042489a9b167e2c82276e0f854255b

See this the two files, one is the log when running, the other is my statefile.

Keridos, 2020-04-09:
Everything works now, thanks! And sorry for the late reply.