sibradzic/amdgpu-clocks #10: Systemd Unit doesn't restart after resume from suspend

terencode, 2020-03-17:
I'm on ArchLinux, I installed it from the AUR (https://aur.archlinux.org/packages/amdgpu-clocks-git/) and I enabled the systemd unit which correctly outputs: 
```
Created symlink /etc/systemd/system/multi-user.target.wants/amdgpu-clocks.service → /usr/lib/systemd/system/amdgpu-clocks.service.
Created symlink /etc/systemd/system/sleep.target.wants/amdgpu-clocks.service → /usr/lib/systemd/system/amdgpu-clocks.service.
Created symlink /etc/systemd/system/suspend.target.wants/amdgpu-clocks.service → /usr/lib/systemd/system/amdgpu-clocks.service.
Created symlink /etc/systemd/system/hibernate.target.wants/amdgpu-clocks.service → /usr/lib/systemd/system/amdgpu-clocks.service.
```

However, after resuming from suspend, the unit is not restarted and thus the different settings not applied.

sibradzic, 2020-03-18:
@terencode thanks for report. Can you please try this in your `amdgpu-clocks.service` file (followed by `systemctl daemon-reload`) and report back:
```
[Unit]
Description=Set custom amdgpu clocks & voltages
After=rc-local.service plymouth-start.service systemd-user-sessions.service suspend.target hibernate.target hybrid-sleep.target

[Service]
Type=oneshot
ExecStart=/usr/bin/amdgpu-clocks
RemainAfterExit=yes
StandardOutput=syslog

[Install]
WantedBy=multi-user.target suspend.target hibernate.target hybrid-sleep.target
```
?

sibradzic, 2020-03-18:
https://askubuntu.com/questions/661715/make-a-script-start-after-suspend-in-ubuntu-15-04-systemd/661747#661747

terencode, 2020-03-18:
It doesn't restart when resuming because of the `RemainAfterExit=true` line.
You have 3 options: https://github.com/rncbc/rtirq/pull/4#issuecomment-452429118

sibradzic, 2020-03-22:
As you can see I've opted for option 3, an additional scriptlet @ `/usr/lib/systemd/system-sleep`. I wanted to keep `RemainAfterExit=true` so that the provided service can be stopped, which will now restore the OverDrive states to the initial values (this is something I planned adding long time ago...).
Please check if this approach works for you.

terencode, 2020-03-22:
Thanks, it works but you should mention in the README that you need to make the script executable (chmod a+x). 

Why did you change the path to /usr/local/bin ?

terencode, 2020-03-22:
Oh and you need to change`/usr/bin/systemctl start amdgpu-clocks` to `/usr/bin/systemctl restart amdgpu-clocks` otherwise it doesn't work.

Also, little typo in your repo description ;) 
[image]

sibradzic, 2020-03-23:
> Thanks, it works but you should mention in the README that you need to make the script executable (chmod a+x).

It is already set as executable in the git, no need to do chmod.

> 
> Why did you change the path to /usr/local/bin ?

It seems more appropriate that way, but it the end of the day it is just a simple script, feel free to place it anywhere you prefer...

sibradzic, 2020-03-23:
> Oh and you need to change`/usr/bin/systemctl start amdgpu-clocks` to `/usr/bin/systemctl restart amdgpu-clocks` otherwise it doesn't work.

Hmm, are you sure, how does it behave for you? It works just fine for me (Ubuntu 20.04) after suspend with just `start`.

> 
> Also, little typo in your repo description ;)
> [image]

Fixed, thanks for pointing it out!

terencode, 2020-03-23:
> It is already set as executable in the git, no need to do chmod.

Ah yeah I needed to chmod it because I mistakenly used `install -Dm 644`

> It seems more appropriate that way, but it the end of the day it is just a simple script, feel free to place it anywhere you prefer...
Yeah, I guess, I was wondering from a packaging perspective.

> Hmm, are you sure, how does it behave for you? It works just fine for me (Ubuntu 20.04) after suspend with just start.

Yes , the unit is not restarted.
You can check this by doing `systemctl status amdgpu-clocks.service`, make sure it's started and take note of the time when it did, then `systemctl start amdgpu-clocks.service` and see the time is the same with  `systemctl status amdgpu-clocks.service` again.

terencode, 2020-03-23:
To quote the manual (https://www.freedesktop.org/software/systemd/man/systemd.service.html): 
> Since the unit is considered to be running after the start action has exited, invoking systemctl start on that unit again will cause no action to be taken.

sibradzic, 2020-03-23:
Ok, changed to restart. Check it out once more, and feel free to close the issue.

closed 2020-03-23 by terencode