sibradzic/amdgpu-clocks #18: line 132: echo: write error: Invalid argument

nikolaos-spyratos, 2020-10-18:
Seems to be when trying to write the `VDDC_CURVE_STATE` values, since I can see the other values I have specified being changed. Nothing seems out of the ordinary with that part of the script so not sure where to look further.

jdotdarko, 2020-10-19:
I had this issue yesterday while I was on 5.9.0-3. Today I updated to 5.9.1-4 and I can now write to pp_od_clk_voltage. 

The issue was mentioned here: https://gitlab.freedesktop.org/drm/amd/-/issues/1245

nikolaos-spyratos, 2020-10-19:
I'm on 5.8.1 but I'll get to trying 5.9.1 then.

alpha6, 2020-10-25:
I have similar problem on my 5700xt/rx580 and 5.8.0-23

jdotdarko, 2020-10-26:
> I have similar problem on my 5700xt/rx580 and 5.8.0-23

Just upgrade to 5.9.1. I'm on RX580 too and I was having the same problem on 5.9.0.

sibradzic, 2020-11-01:
@nikolaos-spyratos & @jdotdarko can you please check against latest fix https://github.com/sibradzic/amdgpu-clocks/commit/5b48f04 ? I finally had time to test 5.10, and there is indeed an issue, as **pp_od_clk_voltage** output changed in the driver, yet again...

jdotdarko, 2020-11-08:
@sibradzic the latest fix works. No issue with it or whatsoever. I'm on 5.9.6

sibradzic, 2020-11-08:
@jdotdarko thanks for reporting!

closed 2020-11-08 by sibradzic