sibradzic/amdgpu-clocks #20: Polaris voltage not working?

FydeNCC, 2020-11-27:
I'm trying to undervolt a few GPUs (Polaris) but the values don't seem to be applied, this is what I'm using:

OD_SCLK:
7: 1150 850
OD_MCLK:
1: 2150 1000
FORCE_SCLK: 7
FORCE_MCLK: 1
FORCE_PERF_LEVEL: manual

(The default value is 1000mV)
The clock speed is applied but the voltage is not, I tested this on RX 470 and RX 580, I also tried to change voltage in all 7 states but there's no difference.
Am I missing something?

sibradzic, 2020-11-27:
AFAIK Polaris never actually supported memory voltage changes via driver interface. Values are there, the API will not reject input values but the changes are simply ignored.

FydeNCC, 2020-11-30:
I see, thanks for your answer.

closed 2020-11-30 by FydeNCC