sibradzic/amdgpu-clocks #26: Can't access /sys/module/amdgpu/parameters/ppfeaturemask on HiveOS

awhorton, 2021-02-20:
I'm not sure this is an issue you'd like to address, but I thought I would submit this in case others have the issue.  On HiveOS I followed your instructions to install the script so that it works on boot, but it was not applying my settings.  I found the error message "Can't access /sys/module/amdgpu/parameters/ppfeaturemask" in syslog despite the file existing.  I was able to resolve this by editing the amdgpu-clocks.service file and adding the following line under the [Service] section.

ExecStartPre=/bin/sleep 30

Probably not the best solution but for a mining rig it's fine.

sibradzic, 2021-02-21:
Yes, stuff under `/sys/module/amdgpu` is present only after the `amdgpu` driver is actually loaded and initialized, which normally happens pretty early in the boot process, which is usually before systemd initializes. But if you have multiple cards, the driver init may take more time. What's your kernel and what type & how may cards does your system have plugged in?

awhorton, 2021-02-21:
Linux 5.4.80-hiveos #201206 SMP x86_64 GNU/Linux

There are six AMD RX 580s in the system.

sibradzic, 2021-02-22:
I see, makes sense, 6 cards will delay the driver init. Can you try this patch (remove `ExecStartPre`):
```
diff --git a/amdgpu-clocks.service b/amdgpu-clocks.service
index 8f8337e..600b014 100644
--- a/amdgpu-clocks.service
+++ b/amdgpu-clocks.service
@@ -1,6 +1,7 @@
 [Unit]
 Description=Set custom amdgpu clocks & voltages
 After=multi-user.target rc-local.service systemd-user-sessions.service
+ConditionPathExists=/sys/module/amdgpu/parameters/ppfeaturemask
 
 [Service]
 Type=oneshot
 ```
 ?

In theory, this should no allow service to start before `ppfeaturemask` exist, but I am not sure if it would just fail the service on first check of `ppfeaturemask`. Try it out please.

awhorton, 2021-02-22:
It just resulted in the service not starting:

● amdgpu-clocks.service - Set custom amdgpu clocks & voltages
   Loaded: loaded (/home/user/amdgpu-clocks/amdgpu-clocks.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
Condition: start condition failed at Mon 2021-02-22 12:03:39 EST; 2min 22s ago
           └─ ConditionPathExists=/sys/module/amdgpu/parameters/ppfeaturemask was not met

sibradzic, 2021-02-22:
Bah, systemd :|
Try adding this as well to the `[Unit]` section:
```
Requires=graphical.target
```
It would probably work, but there could be a system without `graphical.target`...

sibradzic, 2021-03-11:
@awhorton ping

sibradzic, 2021-04-05:
@awhorton any news?

closed 2021-04-08 by sibradzic