sibradzic/amdgpu-clocks #27: Can no longer change ppfeaturemask without crash on startup

Electro-Induced-Original, 2021-02-27:
Hello there,

I have recently been having an issue where enabling amdgpu.ppfeaturemask=0xffffffff in grub causes a crash on start up with heavy artifacting.

I am running Linux Mint with the 5.11 xanmod kernel. I am using a vega 56 flashed to 64, and I had successfully been running it for some time undervolted with a memory overclock.
However, I recently made some changes to my undervolt as it was a little unstable but now, despite changing it back to its original undervolted settings, I get hard crashes where the entire screen is multicoloured artifact. On startup, after decrypting my drive and before logging in as user, it immediately crashes with artifacting. The only way to access my system is to boot in recovery mode and disable amdgpu.ppfeaturemask=0xffffffff in grub and restart the system, so thankfully I can use my system and graphics card but only at default settings.

I think the root issue is an unstable memory configuration. However, I am unable to change this as I am unable to enable amdgpu.ppfeaturemask=0xffffffff without causing crashing. Consequently I cannot run "amdgpu-clocks restore" or change pp_od_clk_voltage.

I was wondering if there was any other way to revert to original clocks in order to start again with my overclocking or if there were any other suggestions as to how I could get around this problem.

Many thanks!

sibradzic, 2021-02-28:
Hi,

That smells like a HW issue. I suggest you disable ppfeaturemask entirely and revert to original vBIOS and try again from scratch.
And, instead of flashing your card with vBIOS meant for some other card & risking breaking the HW, I recommend https://github.com/sibradzic/upp.

Electro-Induced-Original, 2021-03-07:
Hello there,

I ended up "fixing" this by reinstalling linux mint and my drivers which has resolved the issue. Thank you for the advice regarding the vBIOS, I will likely go this route instead.

closed 2021-03-07 by Electro-Induced-Original