sibradzic/amdgpu-clocks #29: cannot access '/etc/default/amdgpu-custom-state*.card*'

kjmann1, 2021-03-22:
Im not sure what im doing wrong but Ive followed the installion process correctly but keep get the error ls: cannot access '/etc/default/amdgpu-custom-state*.card*': No such file or directory

sibradzic, 2021-03-23:
https://github.com/sibradzic/amdgpu-clocks/blob/master/README.md#installing-and-manually-running-the-script
`sudo ls -alh /etc/default/amd*`

sibradzic, 2021-04-08:
ping?

vosian, 2021-04-12:
I have the same issue, there are no amd named files in /etc/default so that returns `no matches found`, perhaps southern island cards don't generate the necessary file for overclocking?

sibradzic, 2021-04-13:
@vosian did you actually check the README? The README sais that **you** are supposed to create `amdgpu-custom-state.cardX` file(s) in `/etc/default` **on your own**. Perhaps the README is unclear in some way?

vosian, 2021-04-13:
While I didn't carefully read the README originally (I scanned through it when I saw the issue and tried to see if I could reproduce the issue), I checked it again, and I couldn't find anywhere that's explicitly stated. We're told where the files are expected to be and roughly what they should contain, but not that their creation depends on the user.

Perhaps, assuming you agree changes would be beneficial, the start of "How does it work" should be changed:

`By default, custom power states are defined in files that are expected to reside in /etc/default/amdgpu-custom-state.cardX`

To begin with I am not 100% sure if this `/etc/default/amdgpu-custom-state.cardX` is a file, or if that's a folder and the files are supposed to go inside said folder, assuming it's the latter. I think the following would work:

`By default, custom power states are stored in the /etc/default/ folder, you have to create a file named amdgpu-custom-state.cardX` and then the rest caould continue as it currently is.

Also, as an aside, while my `/sys/class/drm/cardX/device/pp_od_clk_voltage` file exists (in my case it's card0, the only card I have), it's empty.

sibradzic, 2021-04-13:
> By default, custom power states are defined in files that are expected to reside in `/etc/default/amdgpu-custom-state.cardX`

I thought it is obvious that word "custom" means something that user have to do on her own. But I agree, you are not the first one to misunderstand the meaning, so I will re-word it slightly.

> To begin with I am not 100% sure if this /etc/default/amdgpu-custom-state.cardX is a file, or if that's a folder and the files are supposed to go inside said folder, assuming it's the latter

It is pretty obvious this is a set of files, but I'm about to re-word it so it is 500% obvious.

> Also, as an aside, while my /sys/class/drm/cardX/device/pp_od_clk_voltage file exists (in my case it's card0, the only card I have), it's empty.

That means one of the following two things:
- you haven't set the **ppfeaturemask** correctly, which probably means you haven't read or understood the README
- your GPU driver does not support **pp_od_clk_voltage** interface at all

closed 2021-04-13 by sibradzic

vosian, 2021-04-13:
> That means one of the following two things:
> 
>     * you haven't set the **ppfeaturemask** correctly, which probably means you haven't read or understood the README
> 
>     * your GPU driver does not support **pp_od_clk_voltage** interface at all

I've had 0xffffffff set from the start via kernel parameters.

sibradzic, 2021-04-14:
@vosian hmm, multiple cards?

tonykor, 2021-11-29:
i'm also not understand that i need to create a file, please move the instruction to install section, the same like setting ppfeaturemask to grub

sibradzic, 2021-12-01:
> By default, custom power states for a particular GPU are defined in `/etc/default/amdgpu-custom-state.cardX` file, which is expected to be created by user, and where X corresponds to a card in `/sys/class/drm/cardX`

@tonykor what exactly you did not understand in that sentence?

commit 87f35bf8 (2021-04-13): Clarify custom state file in README (fixes #29)