sibradzic/amdgpu-clocks #35: Write error on line 151 and 165

ThibaultNocchi, 2021-08-22:
Hello,

I'm trying to use your soft on Arch Linux, Linux 5.13.12 on a Ryzen 2200G. I'm trying to run this simple config file:
```
OD_SCLK:
0:        201Mhz
1:       1201Mhz
OD_RANGE:
SCLK:     201MHz       1201MHz
```
with the default system one being at 1101MHz instead of 1201MHz. When running the script, I get this:

[image]

I guess this must be related to me not putting any tension in the config file, and this `power1_cap` file not being found.

What can I do?

Thanks :)

sibradzic, 2021-08-23:
You should read the https://github.com/sibradzic/amdgpu-clocks#readme

ThibaultNocchi, 2021-08-23:
Oh missed it, thanks! Does that mean that I need to supply voltage in my config to make it work, or is there a workaround?

sibradzic, 2021-08-24:
Sorry, the last time I checked the kernel API that manages these "OverDrive" clocks and voltages does not work on APUs like 2200G, only on discrete GPUs.

ThibaultNocchi, 2021-08-24:
Oh too bad ^^ No problem, thanks for your help, have a nice day! :)

closed 2021-08-24 by ThibaultNocchi