sibradzic/amdgpu-clocks #41: About custom states and using restore

GreatBigWhiteWorld, 2022-04-11:
I installed the script as instructed:
`sudo ln -s $(pwd)/amdgpu-clocks /usr/local/bin/amdgpu-clocks`

I however did not put a custom state file under /etc/default, but put one under the directory where I run the script. The reason is that I'd like to make several different custom state files (like profiles).
I test run the script with the following command:
`sudo USER_STATES_PATH=custom-states amdgpu-clocks`
It can run fine with a custom state file under the path where I run the command.
However when I use the restore command I get this:
```
sudo amdgpu-clocks restore
ls: cannot access '/etc/default/amdgpu-custom-state*.card*': No such file or directory

```
Now I am confused and have tons of following questions....

sibradzic, 2022-04-12:
> I test run the script with the following command:
sudo USER_STATES_PATH=custom-states amdgpu-clocks

And what is the name of that directory and a state file?

> I test run the script with the following command:
`sudo USER_STATES_PATH=custom-states amdgpu-clocks`

And the actual output you get is?

> However when I use the restore command I get this:
`sudo amdgpu-clocks restore`
`ls: cannot access '/etc/default/amdgpu-custom-state*.card*': No such file or directory`

TL;DR:
Properly set **USER_STATES_PATH** when using `restore` as well:
`sudo USER_STATES_PATH=custom-states amdgpu-clocks restore`
 
The script default **USER_STATES_PATH** is `/etc/default/amdgpu-custom-state`. You are using custom **USER_STATES_PATH** in the first command, and you probably don't actually have any script-default `/etc/default/amdgpu-custom-state*.card*`files in your system. Since script relies on custom state filename ending (i.e. `.cardX`) to determine the card **drm** index (multiple card support feature), **USER_STATES_PATH** has to point to a file that actually ends with `.card*` (or a directory containing multiple such files).

sibradzic, 2022-04-26:
ping?

GreatBigWhiteWorld, 2022-04-27:
Hello, sorry I had to reinstall the whole system the last time I tried to OC GPU, prolly not directly the fault of this script. It crashed the OS and caused some other issues that can't be fixed without reinstallation. I also upgraded the hardware of the machine a bit. Since then I haven't tried OC again. 
That said, I think you might put in more details about the restore part. Thanks.

closed 2022-04-27 by GreatBigWhiteWorld

sibradzic, 2022-04-27:
Thanks for the update, best of luck!