sibradzic/amdgpu-clocks #45: amdgpu-clocks doesn't automatically apply settings with systemd

spagootie, 2022-07-17:
I'm on Arch Linux x86_64, and the amdgpu-clocks service dowsn't seem to apply automatically when I start my system, or at least something reverts the GPU  to normal states. It works fine if I run `sudo amdgpu-clocks` or `systemctl restart amdgpu-clocks`, but if I log out and log back in, the GPU goes back to normal. I'm running sway with sddm as my login manager.

sibradzic, 2022-07-20:
please share the log output of amdgpu-clocks service following the fresh boot.

spagootie, 2022-07-20:
[amdgpu-clocks.log](https://github.com/sibradzic/amdgpu-clocks/files/9154282/amdgpu-clocks.log)

Here's the log, but it doesn't seem to describe any errors. Like I said, my hypothesis is that starting Wayland messes with the configuration somehow.

sibradzic, 2022-07-22:
First of all, what is the card that you are trying to work with? Do you have multiple cards, or iGPU besides your main card? What about kernel, and amdgpu driver version?

Wayland / Sway should not make any difference, **amdgpu-clocks** script uses kernel sysfs API, DM is not even required for it to work properly. Have you perhaps been using any other AMD overclocking tools in parallel?

> [amdgpu-clocks.log](https://github.com/sibradzic/amdgpu-clocks/files/9154282/amdgpu-clocks.log)

That is just a part of the log I meant. Please share output of
```
journalctl -u amdgpu-clocks.service -b
```
>[amdgpu-custom-states.card0.txt](https://github.com/sibradzic/amdgpu-clocks/files/9154328/amdgpu-custom-states.card0.txt)

Your custom state file is rather "complex". To troubleshoot, you should start with a single simple setting, say just a custom clock/voltage for state 7, and remove everything else.
Oh, **OD_RANGE** is meaningless in custom state file(s), please make sure you had read the README.md in its entirety.

spagootie, 2022-07-22:
I'm running an RX 580 8gb with linux 5.58.12.arch1-1 and mesa 22.1.3-1. I don't have an integrated GPU or other cards. 

I made sure that I wasn't running any other programs like radeon-profile daemon, but now that I think about it I used to have the same exact problem with radeon-profile. 

Here's the log: 
[newlog.log](https://github.com/sibradzic/amdgpu-clocks/files/9172435/newlog.log)

I also commented out every line in my custom state file except for the FORCE_SCLK and FORCE_PERF_LEVEL lines.

sibradzic, 2022-07-23:
According to your log your custom state file is not specifying any changes to the SCLK or MCLK states. Can you try putting just this in the custom state file & paste the amdgpu-clocks service log in the comment:
```
FORCE_PERF_LEVEL: manual
OD_SCLK:
7:       1350MHz       1175mV
```
Following the log please share the output of `cat /sys/class/drm/card0/device/pp_od_clk_voltage`.

sibradzic, 2022-08-06:
@spagootie ping?

closed 2022-09-07 by sibradzic