sibradzic/amdgpu-clocks #49: Power_Cap Navi Silent Bios

dinotheextinct, 2022-11-09:
Hi, I am using this tool for many years now and really appreciate offering something like this.

Sadly I have weird problems with my GPU for years, mostly in graphical intensive games. Where the GPU driver crashes (I assume) and I get a weird black screen with non moving mouse cursor. Need to hard reset or log into a different console to restart the computer.

So now I am tweaking again the amdgpu-custom-states.card0 in hope for a more stable result, previously I was using a power_cap of 180000000 and SCLK of 1890Mhz @ 1050mV.

[image]

Still i got issues, my Radoen RX 5700 XT Red Devil is running on the silent Bios, afaik. So I was curious if the assigned power cap in this tool could cause instability issues? Afaik the silent BIOS caps out at 170V?

I know this is a very hardware specific issue, but most recommendations on the internet are using undervolting stats with windows tools which work a bit different than this, so I was hoping for some feedback on which values I could tweak to achieve more stability.

sibradzic, 2022-11-09:
> curious if the assigned power cap in this tool could cause instability issues?

Well, normally it should not. But then, every card is different, only way to find out is to try it out.

> Afaik the silent BIOS caps out at 170V?

Normally, you can check the default cap using `sensors | grep PPT` command, before applying **amdgpu-clocks** script that is.
If you BIOS defaults at 170W, then there is no need to tweak it in custom state file, you can just comment that out...

dinotheextinct, 2022-11-09:
Hmm with commenting out it shows 180W, interesting. Is there a way to apply changed settings from the amdgpu-custom-states.card0 without having to reboot?

sibradzic, 2022-11-09:
> Is there a way to apply changed settings from the amdgpu-custom-states.card0 without having to reboot?

Yes, just re-run `sudo amdgpu-clocks` or simply restart the service if you are using systemd, after changing the custom state file that is.

closed 2023-01-16 by sibradzic