sibradzic/amdgpu-clocks #51: FORCE_POWER_PROFILE not taking effect

SwooshyCueb, 2023-04-22:
I've set `FORCE_POWER_PROFILE` to `1` in my state file, but after running `amdgpu-clocks`, `pp_power_profile_mode` still reports that `0` is the active profile.
Running with `bash -x`, I can see that `1` does actually get written to `pp_power_profile_mode`. Moving the `FORCE_PROFILE` block to the end of `set_custom_states()` fixes it for me, so I suspect that something happening after `pp_power_profile_mode` is written to is resetting it.

sibradzic, 2023-04-24:
Hi @SwooshyCueb & thanks for reporting!

Hmm, your issue may be likely to the driver quirk or something. Which kernel are you using? Mainline driver or pro?

Also, compare result of moving `FORCE_PROFILE` right before and right after https://github.com/sibradzic/amdgpu-clocks/blob/9d1196ef9839233a6f1b8c343e4769d5229c5015/amdgpu-clocks#L172. The "commit" of updated power state to the driver is known to cause some curious side effects :shrug:

sibradzic, 2023-05-04:
@SwooshyCueb ping?

SwooshyCueb, 2023-05-05:
Apologies, things have been rather busy around here.

I'm using Linux 6.1.7 with some patches, but I don't think any of the patches I use would affect this. I use the regular drivers, not pro.
`
And yeah it looks like the "commit" does reset the power profile reported in sysfs. Moving the `FORCE_PROFILE` block just before `echo 'c' > ${SYS_PP_OD_CLK}` results in it being reset back to default, but it works fine if the block is just after `echo 'c' > ${SYS_PP_OD_CLK}`.

sibradzic, 2023-07-05:
OK, I will re-suffle the code a bit and do some test, expect an updated version soon...

sibradzic, 2023-07-08:
@SwooshyCueb does the latest fix address the issue for you?

SwooshyCueb, 2023-07-10:
Looks like it does! Thank you!

closed 2023-07-14 by sibradzic