sibradzic/amdgpu-clocks #53: Venus XT [Radeon HD 8870M / R9 M270X/M370X]

chumariesco, 2023-06-11:
I'm getting this error after running the script:
'/etc/default/amdgpu-custom-state*.card*': No such file or directory

The boot option was added: amdgpu.ppfeaturemask=0xffffffff

Any idea why this card is not compatible? It is the discrete GPU of a MBP Retina 2015.

Best regards,
Domingo

sibradzic, 2023-07-05:
Hi @chumariesco 

Is this card even using **amdgpu** driver or an old **radeon** driver?

chumariesco, 2023-07-18:
Hi @sibradzic thank you for your reply. I‘ve been for a few weeks away from the machine and will be for the next couple of weeks so, but as far as I know it uses amdgpu.

According to archlinux it uses amdgpu[archlinux](https://wiki.archlinux.org/title/MacBookPro11,x)  with mesa driver:
„ MacbookPro11,5 (MacBook Pro Retina, 15-inch, Mid 2015 with AMD Radeon R9 M370X)
This uses amdgpu through the mesa driver. This should work and be installed by default. The amdgpu-pro drivers do not support this GPU.“

With a GUI called corectrl I‘m able to underclock it to 300MHz, but not sure how it achieves it. Seems profile/governor based. I‘d like to do this (temperature and fan noise control) with your cli, happy to share more details when I‘m back to it.

Kindly,
Domingo

chumariesco, 2023-08-09:
Hi sibradzic! Back to town, I can share now more details on the video card:

$ lspci -v
VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Venus XT [Radeon HD 8870M / R9 M270X/M370X] (rev 83) (prog-if 00 [VGA controller])
        Subsystem: Apple Inc. Radeon R9 M370X Mac Edition
        Flags: bus master, fast devsel, latency 0, IRQ 39
        Memory at 80000000 (64-bit, prefetchable) [size=256M]
        Memory at b0c00000 (64-bit, non-prefetchable) [size=256K]
        I/O ports at 3000 [size=256]
        Expansion ROM at b0c40000 [disabled] [size=128K]
        Capabilities: <access denied>
        Kernel driver in use: radeon
        Kernel modules: radeon, amdgpu

It seems then that it uses an old radeon driver, but it makes use of the amdgpu module somehow? Let me know please if any hints on how to use your script here come to your mind.

sibradzic, 2023-08-22:
Some interesting info at https://www.reddit.com/r/archlinux/comments/vjhk9k/macbookpro115_no_laptop_display_on_reboot_amdgpu/

MegadoomerX, 2023-08-27:
Hi @chumariesco, did you manage to test the script yet? 
 
I am getting same device and was looking into ways to bump graphics performance a little bit. Thanks also for the amdgpu backlight fix link @sibradzic. 
 
The readme just mentions Polaris family cards and upwards, I wasn't able to find anything on GCN1 in regards to overclocking. Is this even possible in general?

chumariesco, 2023-08-28:
Hi @MegadoomerX !

I don't think it's possible to OC. Even with AMDGPU properly driving (thanks for that reference @sibradzic!),  pp_od_clk_voltage is not there. Neither does  "power_dpm_force_performance_level" take 'manual' as profile, which seems a condition for any AMDGPU OC.
/sys/class/drm/card0/device/power_dpm_force_performance_level controls SCLK and accept profiles 'high' (800MHz) and 'low' (300MHz).
/sys/class/drm/card0/device/power_dpm_state controls MCLK and takes 'battery' (300MHz), 'balanced' and 'performance' (1.1GHz).

Personally I downclock it to 'low' and 'battery' to keep the fans running idle and quiet. This machine can get very loud and mostly because of the eGPU temp peaks (not that in OSX it was any quieter). I keep the CPU cores also throttled to 1.6GHz to ensure low temps and power is enough at least for video editing.

sibradzic, 2023-08-28:
So, messing with ppfeaturemask does not expose any controls via **pp_od_clk_voltage** at all?
If not, this may likely be a silicon limitation, and my script will hardly help. Perhaps we should close this issue?

chumariesco, 2023-08-28:
I think so, I'll just leave below the script I'm running at startup (as service) for this MBP. It profiles the GPU to underclock it and fix the backlight issue when AMDGPU installed. Thanks for your patience and guidance anyways!

```
#!/bin/sh -e
echo 1 > /sys/class/backlight/amdgpu_bl0/brightness
echo "battery" | sudo tee  /sys/class/drm/card0/device/power_dpm_state
echo "low" | sudo tee  /sys/class/drm/card0/device/power_dpm_force_performance_level
exit 0
```

closed 2023-08-28 by chumariesco