sibradzic/amdgpu-clocks #55: OD_VDDGFX_OFFSET does not work

hostingnuggets, 2023-06-18:
Hello, I have a Nitro 6700XT GPU on Ubuntu 22.04 LTS (5.15.0-25-generic kernel) and I am trying to undervolt it by -150mV but as far as I understand it does not do anything.

Here is by amdgpu-custom-state file:

```
OD_SCLK:
1: 1800Mhz
OD_MCLK:
1: 1000MHz
OD_VDDGFX_OFFSET:
-150mV
FORCE_PERF_LEVEL: manual
```

When running the `amdgpu-clocks` command there are no error messages and all other parameters get applied correctly (for example the CPU clock) but the undervolting simply does not seem to work.

I checked the `hwmon/hwmon3/in0_input` file of that specific GPU before and after running the script but the value does not change and stays at `843`.

I am using version `22.20.50200` of the amdgpu drivers and applied the `amdgpu.ppfeaturemask=0xffffffff` kernel parameter at boot.

Am I doing something wrong? Or is this a bug maybe?

sibradzic, 2023-07-05:
Hi @hostingnuggets , thanks for reporting.
On my 6600XT here I see no issue with this. I don't think the `hwmon` is not a correct indicated whether a voltage offset is applied or not. What does your `/sys/class/drm/card0/device/pp_od_clk_voltage` (assuming your card is `card0`) say after **amdgpu-clocks** applies the custom state?

hostingnuggets, 2023-07-05:
Hi @sibradzic, thank you very much for taking the time for answering me.
You are totally right I checked the `/sys/class/drm/card1/device/pp_od_clk_voltage` file after running your script and the change is correctly applied as you can see below:

```
OD_VDDGFX_OFFSET:
-150mV
```  
I just thought this change would be reflect in that hwmon file I mention.
My goal is actually to set the core voltage of my GPU to exactly 675mV. Maybe I am using the wrong parameter or even tool for that purpose. Would it maybe be possible with your other tool https://github.com/sibradzic/upp ?

sibradzic, 2023-07-07:
I don't think you can force the voltage to a fixed value. A negative `OD_VDDGFX_OFFSET` will normally decrease the voltage value on the frequency-voltage curve, so that the card would use less power altogether, but it is still going to be a curve, meaning voltage ain't gonna be at fixed value at different frequencies.
Using upp it may be possible to "flatten" the aforementioned curve, but I have never tried that. I think the best course of action in your case would be to use negative voltage offset + some conservative power limit, which would result in your card never exceeding certain voltage threshold.

closed 2023-08-09 by sibradzic