sibradzic/amdgpu-clocks #56: Fails to work with linux 6.4

nutta-git, 2023-08-12:
Hi, 
amdgpu-clocks-git r49.ee457c7-1 does not work with linux 6.4 on archlinux. 

```
sudo amdgpu-clocks
ls: cannot access '/sys/class/drm/card0/device/hwmon': No such file or directory
WARNING: /sys/class/drm/card0/device/pp_od_clk_voltage does not exist, skipping!
```
here is my config file 
```
FORCE_PERF_LEVEL: manual
OD_SCLK:
1: 2900MHz
OD_MCLK:
1: 1000MHz
OD_VDDGFX_OFFSET:
-50mV
```
It does work with linux-lts 6.1, but fails to run with linux and linux-zen.

QuentaceQ, 2023-08-14:
Hello you should check in /sys/class/drm/ for the folder "card" with a number at the end
Then check the file name of your custom state in /etc/default/amdgpu-custom-states.card0
change card0 to what correspond in /sys/class/drm/
In my case i had to change it to amdgpu-custom-states.card**1**

nutta-git, 2023-08-15:
@QuentaceQ 
That seems to be issue, for 6.1 mine was 0 and for 6.4 its 1. 
Thank you

closed 2023-08-15 by nutta-git