sibradzic/amdgpu-clocks #57: Does not work for 10th card

mpmedearis, 2024-03-16:
Works fine for cards 0-9, but will not work for 10th card. I assume the script is just looking at the first number.

sibradzic, 2024-03-18:
@mpmedearis I will be away from keyboard for a while. In a mean time, a PR is more then welcome to fix this.

sibradzic, 2024-03-26:
> Works fine for cards 0-9, but will not work for 10th card. I assume the script is just looking at the first number.

The script does not stop looking at 9, it should parse all `/etc/default/amdgpu-custom-state.card*` files just fine, even if `*` is not a number. So what's happening in your case? How is the 10th card enumerated by kernel driver (`/sys/class/drm/card?/device`)?

Also, if you feel you have too many cards, feel free to send one here :smile

mpmedearis, 2024-03-26:
I think it might have been an error on my part. That was the only rig I had with 11 cards in it, so unfortunately I was unable to test on another rig to verify it was my mistake. Anyway, Thanks a bunch for looking at it.

sibradzic, 2024-03-27:
@mpmedearis cool, but can you please share how is the 10th card enumerated by kernel driver (/sys/class/drm/card?/device)?

sibradzic, 2024-04-11:
I guess not...

closed 2024-04-11 by sibradzic

mpmedearis, 2024-04-24:
sorry for delay.

ls sys/class/drm

card0             card1-DP-4      card3-HDMI-A-5  card7
 renderD129
card0-DP-1        card1-DP-5      card4           card7-DP-19
 renderD130
card0-DP-2        card1-HDMI-A-2  card4-DP-12     card7-DP-20
 renderD131
card0-DP-3        card1-HDMI-A-3  card4-DP-13     card7-DP-21
 renderD132
card0-DVI-D-1     card1-VGA-1     card4-DP-14     card7-DVI-D-6
 renderD133
card0-HDMI-A-1    card2           card4-HDMI-A-6  card7-HDMI-A-9
renderD134
card1             card2-DP-6      card5           card8
 renderD135
card10            card2-DP-7      card5-DP-15     card8-DP-22
 renderD136
card10-DP-26      card2-DP-8      card5-DP-16     card8-DP-23
 renderD137
card10-DP-27      card2-DVI-D-2   card5-DP-17     card8-DP-24
 renderD138
card10-DP-28      card2-HDMI-A-4  card5-DVI-D-4   card8-HDMI-A-10
 renderD139
card10-HDMI-A-12  card3           card5-HDMI-A-7  card9            version
card11            card3-DP-10     card6           card9-DP-25
card11-DP-29      card3-DP-11     card6-DP-18     card9-DVI-D-7
card11-DVI-D-8    card3-DP-9      card6-DVI-D-5   card9-HDMI-A-11
card11-HDMI-A-13  card3-DVI-D-3   card6-HDMI-A-8  renderD128

ls /sys/class/drm/card10
card10-DP-26  card10-DP-28      dev     power      uevent
card10-DP-27  card10-HDMI-A-12  device  subsystem

and of course there is a bunch of other stuff. Let me know what info you
need

sibradzic, 2024-05-12:
Thanks, this is really helpful!