sibradzic/amdgpu-clocks #59: Can't apply anything with amdgpu-clocks

Pacarius, 2024-04-05:
When i ran amdgpu-clocks with the following file:
---------------------------------------------
OD_VDDGFX_OFFSET:
-100mV
----------------------------------------------
I get this::
----------------------------------------------
Detecting the state values at /sys/class/drm/card1/device/pp_od_clk_voltage:
  SCLK state 0: 500Mhz
  SCLK state 1: 2590Mhz
  MCLK state 0: 97Mhz
  MCLK state 1: 1219MHz
  VDD GFX Offset: 0mV
  Maximum clocks & voltages:
    SCLK clock 5000Mhz
    MCLK clock 1500Mhz
  Unexpected value in /sys/class/drm/card1/device/pp_od_clk_voltage:12
-------------------------------------------------------------------------------------------------------------------
and my /sys/class/drm/card1/device/pp_od_clk_voltage looks like this:
--------------------------------------------------------------------------------------------
cat /sys/class/drm/card1/device/pp_od_clk_voltage
OD_SCLK:
0: 500Mhz
1: 2590Mhz
OD_MCLK:
0: 97Mhz
1: 1219MHz
OD_VDDGFX_OFFSET:
0mV
OD_RANGE:
SCLK:     500Mhz       5000Mhz
MCLK:      97Mhz       1500Mhz
VDDGFX_OFFSET:    -450mv          0mv
----------------------------------------------------------------------------
My gpu is a 7800xt hellhound, i am running arch 6.8.1, and i think i am running mesa 24.0.4-2.

sibradzic, 2024-04-07:
Hi @Pacarius, thanks for reporting.

The `VDDGFX_OFFSET: -450mv 0mv` in your **pp_od_clk_voltage** looks super weird and does not look like it belongs there, at least not in the vanilla kernel. For comparison, here is my **pp_od_clk_voltage** (Arch, vanilla kernel *6.8.4-arch1-1*, RX 6600 XT):
```
cat /sys/class/drm/card1/device/pp_od_clk_voltage
OD_SCLK:
0: 700Mhz
1: 2524Mhz
OD_MCLK:
0: 97Mhz
1: 1000MHz
OD_VDDGFX_OFFSET:
-70mV
OD_RANGE:
SCLK:     500Mhz       3150Mhz
MCLK:     674Mhz       1200Mhz
```

You said you are running arch 6.8.1, but your **pp_od_clk_voltage** does not look like a normal vanilla amdgpu driver. Are you perchance running custom or zen kernel or custom driver?

Or maybe the 7800XT HellHound has **PP_OD_FEATURE_GFX_VF_CURVE_BIT** enabled?!? Can you please share your `/sys/class/drm/card1/device/pp_table`? It is binary file, so either encode it in bas64 & dump as a comment or share via some file sharing service...

Pacarius, 2024-04-08:
MBUUAAMAQAN0EAAAfFYAAIUACAAAAB0AAAAAAAB2AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAIMAAAAWAAAAKQAAAAEBAQEBAQEBAQEBAQEAAQAAAQAAAAAAAAAAAAAAAAAAiBMAAIgTAADcBQAA3AUAAA8AAAAcDAAAHAwAAG4AAABuAAAAAQAAAAEAAAABAAAAAQAAAAEAAABkAAAAZAAAAGQAAABkAAAAZAAAAGQAAABkAAAAZAAAAGQAAABkAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAPQBAAD0AQAAYQAAAGEAAAAKAAAA9AEAAPQBAAAZAAAAMgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAGQAAACMAAAAZAAAAIwAAABkAAAAjAAAAGQAAACMAAAAZAAAAIwAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAoAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAGkAXwAAAAAADgakBgAAAAAOBvwIAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAACwAAAD///9xvOsDAAEABADkALAEAAAAAAAAAAAAAAAAAAAAAAAA3ABdACMA3ABdACMAZABuAG4AbgBsAHMAcwBzAHMAcwAAAAAAAAB4AAAAAAAAAAAAAAAAAAAAAAAAAAAAAADy40MA//+bAWQAZABkAGQA+BHAEvgRwBIFAAUANwA3APAK8ArwCvAK8ArwCvAK8AoAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAQAAAGQACgAAAAAAAAAAAAAAAAD2l92+9ijcPWlvcD4AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAIAAAAAAAAAAAAAAAIDAACAPwAAAAAAAAAAAAAAAAAAAAAAAAAA9AEAAAAAAgBiEKg/8KdGPgAAAAAAAAAAAAEEAAAAAAAAAAAAAAEEAKrxwj+mCsY9AAAAAAAAAAAAAAAAAAAAAAAAAAAAAQgDAACAPwAAAAAAAAAAAAAAAAAAAAAAAAAA/AgAAAAAAgB4nII/oBqvPgAAAAAAAAAAAAAAAAAAAAAAAAAAAAACAFCNZz+30UC+AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAIAeJyCP6Aarz4AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAgBQjWc/t9FAvgAAAAAAAAAAAAAAAAAAAAAAAAAAAAACAP7USD/b+T4/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAIA/tRIP9v5Pj8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAgCYbqI/EoNAPgAAAAAAAAAAAAAAAAAAAAAAAAAAAAACAPLSjT97FA4/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAIAmG6iPxKDQD4AAAAAAAAAAPQB0AcAAAAAAAAAAPQBcwoAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAQJ2CwAAAAAAAAAAAAAAAAECmAgAAAAAAAAAAAAAAAAsAdwFYQDJAa0C6AMAAAAAYQDJAQUDwwSUAGYIAAAAAAAAAAAAAAAAlABmCAAAAAAAAAAAAAAAAM0CzQIAAAAAAAAAAAAAAACUABwGAAAAAAAAAAAAAAAAlAAcBgAAAAAAAAAAAAAAAFgC6AOwBEAGCAeeB5gI/AhzCgAA3AUAAMMEAAD8CAAAmAgAAHYLAACYCAAAdgsAAGYIAABmCAAAzQIAABwGAAAcBgAA9AG8AvAKgAwAAAAAIAOQASAD+gABAAEEAQAAABQFAABg6gAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAACAPwAAAABmZmY/awAAAGQAZAAvhBwA4XqUPwAAAACambk/MzOzP9ejcD8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAGQAtgHbAqMEAQrwCgMCAQAAAgMDjArwCrgLgAyIExgVGBUYFQABAgMEBQYHuAu4C7gLuAu4C7gLSA1IDWAJoA/AEgAZIBx4HmAi8CMAAAAAAAEDAQMGTgCcAG8CAAAyADcANwBBADIAQQBBADcANwAAAAAAAAAAADwARgBGAEsAPABLAEsARgBGAAAAAAAAAAAAkAGQAZABkAGQAZABkAGQAZABAAAAAAAAJBMjAEwEuAscDBwM9AHyAwAAAQIAAFMAXwBfAFwARgBkAGQAZABkAAAAAAAAAAAAAAAAAH0AAAB2AHYAdgBxAH0AfQB9AH0AfQAAAAAAAADoA8gALQAcDGQA7P/6AFUAAQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABAAAAAQAAAAEAAAABAAAAAQDgAPAAIAAAAOAA8AAgAAAA4ADwACAAAADgAPAAIAAAAOAA8AAgAAABABMBCwMxAZsDWAEOBIsBAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAPgA+AD4APgA+AAAA//8AAAAAAAEAAAABAAAAAQAAAAEAAAABAOAA8AAgAAAA4ADwACAAAADgAPAAIAAAAOAA8AAgAAAA4ADwACAAAAEAEwELAzEBmwNYAQ4EiwEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA+AD4APgA+AD4AAAD//wAAvAIAALwCAAC8AgAAvAIAALwCAAC8AgAAvAIAALwCAAAAAAAAqgIAAKoCAACqAgAAqgIAAAAAAAAHAAAABwAAAAoAAAAPAAAADwAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAC4CwAARgsAAEYLAABGCwAARgsAAEYLAAAAAAAABwAAAAcAAAAKAAAADwAAAA8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAuAsAAEYLAABGCwAARgsAAEYLAABGCwAAAAAAAGZm5j/NzCxAAABAQAAAYEAAAAAAAAAAAAAAAAAAAAAAAAAAAHcQiz53EIs+VMbvPlTG7z5Uxu8+ZCOwvmQjsL7Ugle/1IJXv9SCV78AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAWAIAALwCAABMBAAAsAQAAAAAAAAYAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAQAAAABVACwB8Ao4Sl49OZdiPJ0R9T7oMF89Yr68ugLxAj/A7B49djcPvpmeED4AAAAAAAAAAAAAAAAAAFUALAHwCjhKXj05l2I8nRH1PugwXz1ivry6AvECP8DsHj12Nw++mZ4QPgAAAAAAAAAAAAAAAAAAVQAsAfAKOEpePTmXYjydEfU+6DBfPWK+vLoC8QI/wOwePXY3D76ZnhA+AAAAAAAAAAAAAAAAbASyPqAVGD0F+iS+bASyPqAVGD0F+iS+bASyPqAVGD0F+iS+AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAQAAAJgAGAAAAJgAAAAgQAAAAAABAAAAmAAYAAAAmAAAAIA/AAAgQQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAACwBCwB0AL0AfQB6AOQAc0ClgIAAAECAQKgAv0BbwKQAbAEvALoAwAAAAAAAJQRAADMEAAAAAAAAAQQAACgDwAAaBAAAAAAAAAAAIAMSA0AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA4wWlCNgJAAAAAAAAAAAAAAAAAAAAAAAAAAAAAPQAAACwBAAAAAAAAAAAAAAwAV0AIwBkAG4AbgBuAGwAcwBzAHMAcwBzAAAAAAAAAAD/ZAD0AYgT9AGIEwAAXgEZAGQAGQBkANQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAM0HAAA+/gAAAAD0AfQBYQBhAPb/9v8jGSMA9AH0ARkAABkAMgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADNBwAAAAAAAAAAiBOIE9wF3AUPAAAAZGRkABwMHAxuAAFkAW4AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAACwBFgC0AL0AfQB6AOQAbwCAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAaBAAAAAAAAAAALwC7gIAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABAAAAuQYAAF8EOy4+BUC8mpkqQc2kw0MAAAAAAAAAAAAAAAAR/G89AABJwQCypUQAAAAAAAAAAAAAAAAAAAAAAAC0
---------------------------------------------------------------------------------------------------------
I don't know if this is correct, i just found the first base64 convertor online and dropped my dumped pp_table in, so if it doesn't work please let me know.
also, I'm at least 99% sure I'm running vanilla arch with vanilla drivers, at least as of rn.

sibradzic, 2024-04-08:
The dump looks good, thanks. I'll be looking into it later today...

Pacarius, 2024-04-09:
looks good! after running the patched script the vddgfx offset properly applies!
----------------------------------Detecting the state values at /sys/class/drm/card1/device/pp_od_clk_voltage:
  SCLK state 0: 500Mhz
  SCLK state 1: 2575Mhz
  MCLK state 0: 97Mhz
  MCLK state 1: 1219MHz
  VDD GFX Offset: 0mV
  Maximum clocks & voltages:
    SCLK clock 5000Mhz
    MCLK clock 1500Mhz
    VDDGFX offset range:    -450mv          0mv
  Curent power cap: 228W
Verifying user state values at /etc/default/amdgpu-custom-state.card1:
  VDD GFX Offset: -100mV
Committing custom states to /sys/class/drm/card1/device/pp_od_clk_voltage:
  Done
[....@.... amdgpu-clocks]$ cat /sys/class/drm/card1/device/pp_od_clk_voltage
OD_SCLK:
0: 500Mhz
1: 2575Mhz
OD_MCLK:
0: 97Mhz
1: 1219MHz
OD_VDDGFX_OFFSET:
-100mV
OD_RANGE:
SCLK:     500Mhz       5000Mhz
MCLK:      97Mhz       1500Mhz
VDDGFX_OFFSET:    -450mv          0mv
----------------------------
Thx <3

commit 327a4e34 (2024-04-10): Handle RDNA3 VDDGFX_OFFSET (fix #59)

This adds parsing for VDDGFX_OFFSET found in pp_od_clk_voltage on RDNA3
cards, and just informs the user about the voltage offset limits.
closed 2024-04-10 by sibradzic