sibradzic/amdgpu-clocks #60: Auto card discovery

husam212, 2024-04-24:
This is a feature request, not a bug.

The card index `cardX` is expected to change sometimes with kernel updates, there's only one card in the system in most use cases. It would be nice if it automatically detects the card and apply a non-indexed config file `/etc/default/amdgpu-custom-states`.

To prevent applying incorrect config, maybe this detection should only work if there's a single card and a non-indexed config, otherwise it falls back to manual configuration.

sibradzic, 2024-04-27:
Hi @husam212, thanks for your feedback.

I totally understand, it is pretty frustrating when the same card pops up with a different identifier. Especially with prevalence of onboard iGPUs, I've personally been hit with the issue of my dGPU `cardX` identifier being randomly assigned on each reboot.

However, something has changed with kernel 6.7 (or was it 6.8?), and now my iGPU always gets assigned with `card0` and dGPU always ends up assigned with `card1`. Heck, even when iGPU is completely disabled in BIOS, my dGPU gets `card1`. I was too lazy to dig through the kernel commits to check exactly why is this the case, but seemingly it just works....

So, before I consider some possible workarounds in this very project; How exactly does the kernel enumerates your card? Which distro & kernel versions have you tried, and which kernel versions changed the enumeration behavior for you?

sibradzic, 2024-05-04:
@husam212 ping?

husam212, 2024-05-04:
Hi @sibradzic sorry I missed your reply notification.

I have one dGPU only, the kernel used to assign it `card0` and after some update it started assigning it `card1`. My use case is undervolting and downclocking to lower temps, at some point I figured the temps are higher than usual, that lead me to discover that's no longer `card0`, I don't really know at which version update it flipped to `card1`. The kernel I'm using now is 6.8.7.

For now I added a symlink for `card1` in case the kernel decided to assign `card0` for some reason.

sibradzic, 2024-05-05:
Yeah, same thing here, dGPU only (iGPU disabled in BIOS), was `card0` and now is consistently `card1`. Let's see if anything changes with kernel 6.9 and address this accordingly.

In the meantime, I've worked out this patch that allows specifying custom state file with a card's PCI address suffix in its name, for example: `/etc/default/amdgpu-custom-state.pci:0000:03:00.1`. This way the custom state should always point to correct `cardX`, no matter how it gets enumerated by kernel on boot. Feel free to test it out:
```diff
diff --git a/amdgpu-clocks b/amdgpu-clocks
index b26fa1f..b2e145e 100755
--- a/amdgpu-clocks
+++ b/amdgpu-clocks
@@ -217,10 +217,15 @@ function clear_vars() {
 
 check_ppfeaturemask
 
-for USER_STATE_FILE in "${USER_STATES_PATH}"*.card* ; do
+#TODO: update README with example about /etc/default/amdgpu-custom-state.pci:0000:03:00.1
+for USER_STATE_FILE in "${USER_STATES_PATH}"*.card* "${USER_STATES_PATH}"*.pci* ; do
   BACKUP_STATE_FILE=/tmp/${USER_STATE_FILE##*/}.initial
 
-  sys_device=/sys/class/drm/${USER_STATE_FILE##*.}/device
+  if [[ "${USER_STATE_FILE}" == *"pci"* ]]; then
+    sys_device=$(dirname /sys/class/drm/card*/device/consumer:${USER_STATE_FILE#*.})
+  else
+    sys_device=/sys/class/drm/${USER_STATE_FILE##*.}/device
+  fi
   sys_hwmons=( "${sys_device}"/hwmon/hwmon[0-9] )
 
   SYS_PP_OD_CLK=${sys_device}/pp_od_clk_voltage
```

husam212, 2024-05-05:
I checked my laptop which has a single Intel iGPU, it's assigned `card1` too, so I think the kernel is not consistent in indexing GPUs.

The patch sounds like a good simple fix, I will give it a try.

The patch would still break if the GPU is moved to another PCI slot, right? I was thinking maybe the combination of `device/vendor` and `device/device` would be globally unique identifier for the GPU. That's probably a more complicated fix though.

husam212, 2024-05-11:
I've been using the patch for a while now, it's working great. I guess the PCI address method solves the issue in the majority of cases. Thanks.

Will this patch get merged into `master`?

commit 2f99858b (2024-05-13): Custom state file with .pci.* suffuix (fix: #60)

This will cause custom state to stick to a particular PCI device.
For example, a custom state defined with
/etc/default/amdgpu-custom-state.pci:0000:03:00.0
will only affect PCI card assagned at 0000:03:00.0
closed 2024-05-13 by sibradzic