sibradzic/amdgpu-clocks #70: Applying state seems to have no effect?

byte-bandit, 2025-03-23:
Hi,

I'm running a 9070XT OC on `Linux 6.13` and using the current master of this script, trying to reduce the max boost clock to prevent driver timeout issues.

```bash
❯ cat /etc/default/amdgpu-custom-state.card2 
OD_SCLK_OFFSET:
1: 500Mhz
OD_MCLK:
0: 97Mhz
1: 1259MHz
OD_VDDGFX_OFFSET:
0mV
#FORCE_POWER_CAP: 25000000
FORCE_PERF_LEVEL: manual
# compute
# FORCE_POWER_PROFILE: 5
```

Applying the configuration seems to work fine, but when trying to verify the state file, I cannot see any changes?

```bash
❯ sudo amdgpu-clocks
Won't write initial state to /tmp/amdgpu-custom-state.card2.initial, it already exists.
Detecting the state values at /sys/class/drm/card2/device/pp_od_clk_voltage:
  SCLK offset 0: -500Mhz
  SCLK offset 1: 1000Mhz
  MCLK state 0: 97Mhz
  MCLK state 1: 1259MHz
  VDD GFX Offset: 0mV
  Value ranges:
    SCLK offset 1000Mhz
    MCLK clock 1500Mhz
    VDDGFX offset range:    -200mv          0mv
  Current power cap: 317W
Verifying user state values at /etc/default/amdgpu-custom-state.card2:
  SCLK offset 1: 500Mhz
  MCLK state 0: 97Mhz
  MCLK state 1: 1259MHz
  VDD GFX Offset: 0mV
  Force performance level to manual
Committing custom states to /sys/class/drm/card2/device/pp_od_clk_voltage:
Skipping ZERO_RPM_ENABLE: make sure to have Linux 6.13 or newer.
Skipping FAN_ZERO_RPM_STOP_TEMPERATURE: make sure to have Linux 6.13 or newer.
  Done
~ 
❯ cat /sys/class/drm/card2/device/pp_od_clk_voltage
OD_SCLK_OFFSET:
0: -500Mhz
1: 1000Mhz
OD_MCLK:
0: 97Mhz
1: 1259MHz
OD_VDDGFX_OFFSET:
0mV
OD_RANGE:
SCLK_OFFSET:    -500Mhz       1000Mhz
MCLK:      97Mhz       1500Mhz
VDDGFX_OFFSET:    -200mv          0mv
```

Any idea on what's going on here is appreciated.

Lawstorant, 2025-03-24:
Overdrive was not implemented properly for RDNA4 in 6.13. I sent a patch for amdgpu that changes the offset handling and it's now merged for 6.14.
https://gitlab.freedesktop.org/drm/amd/-/issues/4036

byte-bandit, 2025-03-25:
Appreciate your effort and explanations, thank you! The values did seem weird to me, since I couldn't see the official boost clock of 2940Mhz anywhere. It's an OC card, so I had assumed that the `1000Mhz` were just blindly slapped on top of the regular `2400Mhz` clock, which seemed insane.

Closing this for now until `6.14` lands, it's already in testing :)

closed 2025-03-25 by byte-bandit

Lawstorant, 2025-03-25:
Actually, I just got notified that Greg applied these patches to 6.12 and 6.13 stable trees So both LTS and 6.13.9 will have them.

Zero RPM accepted into amdgpu-drm-next and I suppose it might be backported as well.