sibradzic/amdgpu-clocks #73: RDNA3 - Applying settings is not reliable

madness742, 2025-06-06:
## Description
Applying settings on RDNA3 GPUs has been very unreliable, it will often fail to adjust the maximum clock and requires changing it to another value before it takes effect.

`LACT` has implemented workarounds for RDNA3 with version 0.7.3, which has solved the issue with my GPU. 

Would it be possible to implement something similar in `amdgpu-clocks`?

## Relevant links
- https://gitlab.freedesktop.org/drm/amd/-/issues/3625
- https://github.com/ilya-zlobintsev/LACT/issues/520

## /etc/default/amdgpu-custom-state.card1
```
OD_SCLK:
1: 2500Mhz
FAN_ZERO_RPM_ENABLE:
0
```

## System info
Distro: Fedora 42 KDE
Kernel: 6.14.9
GPU: AMD Sapphire NITRO+ RX 7900 XTX

## Systemctl status amdgpu-clocks
```
● amdgpu-clocks.service - Set custom amdgpu clocks & voltages
     Loaded: loaded (/etc/systemd/system/amdgpu-clocks.service; enabled; preset: disabled)
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf
     Active: active (exited) since Fri 2025-06-06 04:53:38 CEST; 16min ago
 Invocation: 7616875532b74790a1331320be38ad92
    Process: 2180 ExecStart=/usr/local/bin/amdgpu-clocks (code=exited, status=0/SUCCESS)
   Main PID: 2180 (code=exited, status=0/SUCCESS)
   Mem peak: 6M
        CPU: 19ms

jun 06 04:53:38 fedora amdgpu-clocks[2180]: Detecting the state values at /sys/class/drm/card1/device/gpu_od/fan_ctrl/fan_zero_rpm_stop_temperature:
jun 06 04:53:38 fedora amdgpu-clocks[2180]:   Zero RPM temperature: 55
jun 06 04:53:38 fedora amdgpu-clocks[2180]:   Value ranges:
jun 06 04:53:38 fedora amdgpu-clocks[2180]:     Zero RPM stop temperature range: 25 100
jun 06 04:53:38 fedora amdgpu-clocks[2180]: Verifying user state values at /etc/default/amdgpu-custom-state.card1:
jun 06 04:53:38 fedora amdgpu-clocks[2180]:   SCLK state 1: 2500Mhz
jun 06 04:53:38 fedora amdgpu-clocks[2180]:   Zero RPM enable: 0
jun 06 04:53:38 fedora amdgpu-clocks[2180]: Committing custom states to /sys/class/drm/card1/device/pp_od_clk_voltage:
jun 06 04:53:38 fedora amdgpu-clocks[2180]:   Done
jun 06 04:53:38 fedora systemd[1]: Finished amdgpu-clocks.service - Set custom amdgpu clocks & voltages.
```

madness742, 2025-06-16:
Hi!

This is the commit in LACT that implements the workaround: https://github.com/ilya-zlobintsev/LACT/commit/63d6adc21e672882d5c3eabe3cc1663db40526c1

As for the order that gets around this issue, I copied the `manual` settings from the second relevant link in my first post and adjusted it to my liking.

```bash
echo "r" > /sys/class/drm/card1/device/gpu_od/fan_ctrl/fan_curve
echo "r" > /sys/class/drm/card1/device/pp_od_clk_voltage

echo "auto" > /sys/class/drm/card1/device/power_dpm_force_performance_level
echo "manual" > /sys/class/drm/card1/device/power_dpm_force_performance_level

echo "s 1 2500" > /sys/class/drm/card1/device/pp_od_clk_voltage

echo "1" > /sys/class/drm/card1/device/pp_power_profile_mode
echo "0" > /sys/class/drm/card1/device/gpu_od/fan_ctrl/fan_zero_rpm_enable

echo "c" > /sys/class/drm/card1/device/pp_od_clk_voltage
echo "c" > /sys/class/drm/card1/device/gpu_od/fan_ctrl/fan_zero_rpm_enable
```

I haven't had any issues with settings not applying after I've made this change. The last two lines are swapped compared to the post where I got these settings from, but I can't remember why. 

Happy to test future builds of `amdgpu-clocks`, if needed! :)

sibradzic, 2025-06-18:
Ok, looks like the https://github.com/ilya-zlobintsev/LACT/commit/63d6adc21e672882d5c3eabe3cc1663db40526c1, contains several changes, some of which may be irrelevant there, so let's go together through what's what:
1. **feat: WIP rdna3 order fixes** - I think this is the only thing that matters for this very bug. The example of a correctly working order is at https://github.com/ilya-zlobintsev/LACT/blob/63d6adc21e672882d5c3eabe3cc1663db40526c1/lact-daemon/src/tests/snapshots/lact_daemon__tests__apply_config__amd__rx7900xtx.snap ?
2. **chore: disable clockdown workaround** - I see that the patch disables previous workaround, probably for the very same issue.
3. **fix: temporarily disable opensuse build** - unrelated
4. **fix: reset fan control at the very start** - I do not really understand if fan control is related to this very bug or not. Since I do not have the card to test myself, can you please confirm if it is related or not? IMHO, if the fan control does not affect the reliability of setting the SCLK and other GPU controls unrelated to fans, we should leave the fan control aside for the time being and figure out the ordering workaround first.
5. **chore: cleanup** - unrelated

Along what I've said in 1. and 4., is this supposed to be correct order when dealing with RX7x00 cards:
1. reset **pp_od_clk_voltage**
2. set voltage/clocks/offsets in **pp_od_clk_voltage**
3. set **power_dpm_force_performance_level** to manual. I see that in your example above you set it twice, first to "auto" and then "manual". Is that required? Does final mode have to be *manual* level or other levels can be applied as final state?
4. set **pp_power_profile_mode** to 5 (LACT test code example), or is it 1 (your example)? I need to know what those numbers represent, so can you please share the output of `cat /sys/class/drm/card1/device/pp_power_profile_mode` for your card?
5. Finally, commit all changes to **pp_od_clk_voltage**

madness742, 2025-06-25:
Correct, the only way I can get it to work is by commenting out the following line.

https://github.com/sibradzic/amdgpu-clocks/commit/4aac56a323b55e051f0fc1de25e8a71a9054f4da#diff-0b7b88ad999987c4f05750225c2ecf4c0a0f5da234da3b61832498747cf50a85R245

I have tried moving it around, but with no success.

yuheho7749, 2025-06-27:
I have a similar issue https://github.com/sibradzic/amdgpu-clocks/pull/63#issuecomment-2742189151. From my testing on RX 7700xt, increasing the power cap reset changes to the OD_SCLK. Also, multiple commits (writing "c" to some settings) also resets other settings that were previously committed. Turns out you only need to commit once at the end. Also, the changes I made for rdna 3 seems to break other gpus, so I ended making my own script instead.

sibradzic, 2025-06-29:
> Correct, it doesn't make any changes now. The only way I can get it to work is by commenting out the following line.
> 
> [4aac56a#diff-0b7b88ad999987c4f05750225c2ecf4c0a0f5da234da3b61832498747cf50a85R245](https://github.com/sibradzic/amdgpu-clocks/commit/4aac56a323b55e051f0fc1de25e8a71a9054f4da#diff-0b7b88ad999987c4f05750225c2ecf4c0a0f5da234da3b61832498747cf50a85R245)
> 
> I have tried moving it around, but with no success.

Wait, are you saying that commenting out `echo 'r' > "$SYS_PP_OD_CLK"` makes everything work for you?
That would be strange, because it goes against what you said @ https://github.com/sibradzic/amdgpu-clocks/issues/73#issuecomment-2978074890

That being said, why don't we first eliminate all FAN manipulation (including reset) from the equation, and just confirm the proper order and all steps that results in `OD_SCLK` being properly set? Can you please remove all FAN config options from your config, leave only the absolute minimum config for just setting `OD_SCLK`, and retest (with full power cycle between applying changes, as SMU FW on some cards could get into broken state after some changes)?

sibradzic, 2025-06-29:
> I have a similar issue [#63 (comment)](https://github.com/sibradzic/amdgpu-clocks/pull/63#issuecomment-2742189151). From my testing on RX 7700xt, increasing the power cap reset changes to the OD_SCLK. Also, multiple commits (writing "c" to some settings) also resets other settings that were previously committed. Turns out you only need to commit once at the end. Also, the changes I made for rdna 3 seems to break other gpus, so I ended making my own script instead.

Thanks for joining in this discussion @yuheho7749, very helpful. So, in your case, increasing `FORCE_POWER_CAP` above HW default resets the `OD_SLCK` changes to HW defaults, but setting `FORCE_POWER_CAP` below HW default does not reset `OD_SLCK` changes?

About multiple `OD_SLCK` commits (`echo 'c' > "$SYS_PP_OD_CLK"`) , **amdgpu-clocks** never does that. It commits only once, after all `OD_SLCK` changes have been accounted for, by design. Multiple commits caused odd issues on some older generation hardware, figured out that it should be avoided long ago.

Care to share your custom script, so we can have better understanding of _working_ order of chaining things @ RX7700XT?

yuheho7749, 2025-06-29:
> Thanks for joining in this discussion @yuheho7749, very helpful. So, in your case, increasing FORCE_POWER_CAP above HW default resets the OD_SLCK changes to HW defaults, but setting FORCE_POWER_CAP below HW default does not reset OD_SLCK changes?

Yes. I couldn't figure out why it's so weird.

> About multiple OD_SLCK commits (echo 'c' > "$SYS_PP_OD_CLK") , amdgpu-clocks never does that. It commits only once, after all OD_SLCK changes have been accounted for, by design. Multiple commits caused odd issues on some older generation hardware, figured out that it should be avoided long ago.

I am referring to if I have both changes to `pp_od_clk_voltage` and `gpu_od/fan_ctrl/fan_curve`. I only need to commit to one of the file at the very end to apply both settings. The way your script works might have both commits (`echo "c"`) to `pp_od_clk_voltage` and `gpu_od/fan_ctrl/fan_curve` and that messes things up.
This comment also observe the same behavior: https://github.com/ilya-zlobintsev/LACT/issues/520#issuecomment-2779388009

sibradzic, 2025-07-06:
> Yes. I couldn't figure out why it's so weird, but that's also why I tried to move changes to the POWER_CAP before modifying OD_SCLK and OD_MCLK in that PR.

Tried and succeeded? Have you figured out a correct order needed to successfully set _both_ FORCE_POWER_CAP > HW default & applying changes to OD_SLCK?

> I am referring to if I have both changes to pp_od_clk_voltage and gpu_od/fan_ctrl/fan_curve. 
  . . .
This comment also observe the same behavior: https://github.com/ilya-zlobintsev/LACT/issues/520#issuecomment-2779388009

I see, so this is not about multiple commits to `pp_od_clk_voltage`, but about making commit to both `pp_od_clk_voltage` & `fan_zero_rpm_enable`. This is actually the reason why I asked @madness742 to eliminate FAN changes from the "correct order" test, but still waiting for confirmation. Any news @madness742 ?

Now, since there are 3 different possible commit target sysfs files covered in the script:
- pp_od_clk_voltage
- gpu_od/fan_ctrl/fan_zero_rpm_enable
- gpu_od/fan_ctrl/fan_zero_rpm_stop_temperature (I do not see this one @ RDNA4, can you confirm if this exists on RDNA3 on the latest kernel?)

Is it enough to commit to just one of the 3 to commit all the changes? Like can we commit just only to `pp_od_clk_voltage` to apply FAN 0 RPM and stop temp user changes?

yuheho7749, 2025-07-06:
> Tried and succeeded? Have you figured out a correct order needed to successfully set both FORCE_POWER_CAP > HW default & applying changes to OD_SLCK?

Yes. I just need to set the `power1_cap` before applying the clock speeds and (most) other settings.

> Is it enough to commit to just one of the 3 to commit all the changes?

Yes. I chose to commit to `pp_od_clk_voltage` on my script and it works on all 3 settings. Btw, RDNA4 also seems to work the same way with 1 commit (although `fan_zero_rpm_stop_temperature` does seem to be missing)

> I do not see this one @ RDNA4, can you confirm if this exists on RDNA3 on the latest kernel?)

Sadly, I literally just traded in my 7700xt, so I can't confirm 100%, but I think it still has it last time I checked (a few days ago?) on Arch Linux kernel 6.15.4.

madness742, 2025-07-06:
> Wait, are you saying that commenting out `echo 'r' > "$SYS_PP_OD_CLK"` makes everything work for you? That would be strange, because it goes against what you said @ [#73 (comment)](https://github.com/sibradzic/amdgpu-clocks/issues/73#issuecomment-2978074890)

The `RDNA3-ordering-workaround` branch does not allow me to adjust clock speeds, when commenting out the line I mentioned, the behaviour returns to before (applying settings being unreliable).

> That being said, why don't we first eliminate all FAN manipulation (including reset) from the equation, and just confirm the proper order and all steps that results in `OD_SCLK` being properly set? [...]

I have removed all FAN configuration and only set `OD_SCLK`, with a full power cycle, and it unfortunately still doesn't adjust the clock speeds.

madness742, 2025-07-06:
After commenting out `echo 'r' > "$SYS_PP_OD_CLK"` (Line 245) and  commits to gpu_od/fan_ctrl/fan_zero_rpm_enable & gpu_od/fan_ctrl/fan_zero_rpm_stop_temperature (Line 291 - 306) the clock speed seems to get adjusted properly.

I have tested a full power cycle and couple reboots, and it sticks to the value I specified in `/etc/default/amdgpu-custom-state.card1`.

## cat /etc/default/amdgpu-custom-state.card1
```
OD_SCLK:
1: 2500Mhz
```

sibradzic, 2025-07-06:
Ok, I've updated the https://github.com/sibradzic/amdgpu-clocks/tree/RDNA3-ordering-workaround branch with the following changes:
1. Removed the `echo 'r' > "$SYS_PP_OD_CLK"`
2. Shift power level/profile/cap changes to be executed before any `pp_od_clk_voltage` stuff, as suggested by @yuheho7749 
3. Removed commits to `gpu_od/fan_ctrl/fan_zero_rpm_*`
4. Moved commit to `pp_od_clk_voltage` to the bottom, so that it picks up FAN changes if any (good news is that it works @ RDNA4)

@madness742, please test this, in two rounds, first without any custom FAN options in amdgpu-custom-state.card1 and then with adding custom FAN options.

madness742, 2025-07-06:
Thank you for the changes!

It seems to work properly now, I've also tested the compute performance and it performs how it should. LACT was giving me much lower performance for some reason, as mentioned in https://github.com/sibradzic/amdgpu-clocks/issues/73#issuecomment-2983419366.

The custom settings are persistent across many reboots. I also briefly tested putting the system into suspend, and waking it up (https://github.com/sibradzic/amdgpu-clocks/blob/master/amdgpu-clocks-resume). 

### Without FAN configuration
```
OD_SCLK:
1: 2500Mhz
```

### With FAN configuration
```
OD_SCLK:
1: 2500Mhz
FAN_ZERO_RPM_ENABLE:
0
```

### With FAN configuration + `FORCE_PERF_LEVEL` set to `manual` for ~15% increase in compute performance
```
OD_SCLK:
1: 2500Mhz
FAN_ZERO_RPM_ENABLE:
0
FORCE_PERF_LEVEL: manual
```

sibradzic, 2025-07-06:
Awesome, thanks for testing @madness742. I will have to do a quick regression test against the fix branch using RDNA2 card I have lying around somewhere, and if all goes good it should be merged to main.

closed 2025-07-08 by sibradzic